flynneva/grbl_ros #106: Unlock/ClearAlarm grbl

liushuya7, 2021-03-09:
Hi Evan,

Thanks for creating this package.
I was able to connect to my grbl. However, I could nothing more from that point on. My speculation is that I need to unlock grbl first since grbl initializes in Alarm state. Although I found clearAlarm(), I didn't see any ros service/action to trigger that callback.

flynneva, 2021-03-09:
hi @liushuya7, yeah unfortunately i havent added a specific callback/service for that yet. but you can send the `unlock` command to it using the `send` service provided, just use what is shown in the `clearAlarm` function:

`$X` i think

flynneva, 2021-03-09:
i havent decided what the best way to do this is since it is hard to determine if the user wants to unlock the machine or not.

liushuya7, 2021-03-09:
I tried calling service that way first with `ros2 action info /cnc_001/send_gcode_cmd '{command: $X}'` then with `ros2 service call  /cnc_001/unlock grbl_msgs/srv/Unlock` with my newly defined service type `Unlock`, but somehow it stuck at service_request, nothing seems happening.

liushuya7, 2021-03-09:
Good point, thanks! I will try that later

liushuya7, 2021-03-11:
Hi @flynneva ,

I tried `ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: \$X}'`. Now on the sending terminal I got:
```
Waiting for an action server to become available...
Sending goal:
     command: \$X

Goal accepted with ID: f64ce19ca8c945c39f8c455763f18b7a

Result:
    success: false

Goal finished with status: ABORTED
```
On the grbl node terminal, I got:
```
[grbl_node] [ERROR] [1615442045.956277731] [cnc_001]: Error raised in execute callback: 'grbl_node' object has no attribute 'decode_error'
[grbl_node] [WARN] [1615442045.956782143] [cnc_001]: Goal state not set, assuming aborted. Goal ID: [ 44 116 242 153 159 145  66 169 190 187 147  48   0 197  21 226]
[grbl_node] [INFO] [1615442060.117807715] [cnc_001]: [ \$X ] error:9
```

flynneva, 2021-03-12:
Just updated the README with some instructions on how to clear the alarm. Maybe check it out and let me know if that answers your question?

flynneva, 2021-03-12:
Also maybe double check your parameters in the grbl_ros/config/cnc001.yaml file. I updated it so now to be more generic for whatever you call your node

liushuya7, 2021-03-14:
Great, thanks!

flynneva, 2021-11-05:
@liushuya7 were you able to solve your problem? I can add this to my "todos" since I plan on picking this package up again in over the next few weeks

liushuya7, 2021-11-05:
Hi @flynneva ,
Sorry, I haven't been using your package. I created [grbl_ros2_gui](https://github.com/liushuya7/grbl_ros2_gui.git), which based on [cn5X](https://github.com/fra589/cn5X). But it definitely would be interesting to combine our code to provide the best performance.
Joshua

closed 2021-11-27 by flynneva