flynneva/grbl_ros #107: How to change the pose?

apanagar, 2021-05-15:
I was looking at this package, looks really interesting. I saw the following code at device.py:153 but can't find moveTo defined anywhere or poseCallback being used. using the `send` functionality to send gcode looks simple enough, but is that the intended way to move/change the position of the machine using this project?

```
    def poseCallback(self, request, response):
        self.machine.moveTo(request.position.x,
                            request.position.y,
                            request.position.z,
                            blockUntilComplete=True)
        return response
```

Thanks, awesome work!

flynneva, 2021-05-15:
hey @apanagar! good find! it looks like i mustve wrote that function without ever really implementing it. ill look into this over the weekend and see what I can come up with.

flynneva, 2021-05-15:
@apanagar  so yeah I think when I first started writing this i had the idea of having a callback method where you could pass in an easy "X Y Z" coordinate and the machine would go there. I quickly realized it would just be easier to handle that in the `sendGcode` function since it really does the same thing....so i think i gave up on that `poseCallback`. probably should do some spring cleaning it looks like :)

flynneva, 2021-05-15:
again, I wrote this mostly for fun and to learn ROS2 multithreading actions. i havent used my grbl cnc in quite some time, but maybe this weekend I should pick this one up again and see what is missing.

apanagar, 2021-05-15:
No problem, I'm just using grbl to control a different type of gantry style robot and found your code. I need to integrate grbl with Ros to interface with other subsystems and your code has been super helpful! Thanks for the quick reply!

flynneva, 2021-11-05:
@apanagar did you end up solving your problem? I am about to start some work on this package again and wanted to check in and see if there are still any open bugs

apanagar, 2021-11-12:
I did solve the problem, but my solution was much more crude because I was just getting a prototype to work. Your code was helpful though! 👍

closed 2021-11-27 by flynneva