flynneva/grbl_ros #109: failed to parse global arguments

yagizzha, 2021-11-15:
[image]

i get this error when i try to : ros2 run grbl_ros grbl_node --ros-args --params-file /path/to/workspace/src/grbl_ros/config/cnc001.yaml

flynneva, 2021-11-15:
@yagizzha this error looks like it might be caused by your yaml file having a typo in it.

Could you share your yaml file? Fill it in with fake / dummy data if you'd like but keep the formatting the same.

yagizzha, 2021-11-15:
/**:
  ros__parameters:
    machine_id: cnc_001
    port: /dev/ttyUSB0
    baudrate: 115200
    x_max: 300  # mm
    y_max: 200  # mm
    z_max: 150  # mm
    default_v: 100  # mm / min
    x_max_v: 150  # mm / min
    y_max_v: 150  # mm / min
    z_max_v: 150  # mm / min
    x_steps: 100  # steps / mm
    y_steps: 100  # steps / mm
    z_steps: 100  # steps / mm

yagizzha, 2021-11-15:
didnt change anything in it

flynneva, 2021-11-15:
It has to be the path to where the yaml file is in your system

flynneva, 2021-11-16:
@yagizzha was your issue fixed? are you still seeing the same problem?

yagizzha, 2021-11-16:
[image]
solved one problem but now i get this , do i need to have a working grbl device to start it ?

closed 2021-11-23 by yagizzha