flynneva/grbl_ros #116: GRBL status warning

yagizzha, 2021-12-01:
[image]
getting this warning every time i send a command , takes 2 of same command to move to place i want , also 
[image]
cant echo the position 

what can be causing this ? is it possible for grbl version to be the problem ?

flynneva, 2021-12-01:
What version of grbl do you have?

yagizzha, 2021-12-03:
i am using 1.1 version

flynneva, 2021-12-04:
@yagizzha so that is the same as what I have locally and I've tested this package with that `grbl` version.

Have you used your `grbl` CNC before with another program like [openbuilds-control](https://openbuildspartstore.com/openbuilds-control/) and confirmed your device is functional?

Also, are you following the `README` instructions to first unlock and then run a gcode command?

yagizzha, 2021-12-07:
i havent used it with anything else , tho it gives expected results when i put commands through arduino . i can send gcode commands but what i receive back in ros seems to not be what it expects

closed 2022-02-02 by yagizzha