flynneva/grbl_ros #117: stuck on "Waiting for an action server to become available..."

yagizzha, 2022-02-02:
[image]
getting some errors on machine state , dont know if its related , 
when i try to use "ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: $X}'" i get stuck at 
"Waiting for an action server to become available..."
[image]
Also when i check topics i dont see any grbl ros topics 
[image]

flynneva, 2022-02-02:
@yagizzha thank you for reporting your problem and including the screenshots! This helps tremendously when trying to troubleshoot. From the screenshots it looks like the response from your machine is not in the expected format....except there is a bug in the code that doesnt print out what the response is. Give me one second and I will make a branch for you to test again and hopefully it will print out what your device response is.

flynneva, 2022-02-02:
@yagizzha could you go ahead and try this branch out? https://github.com/flynneva/grbl_ros/tree/117-fix-status-print

I think we had forgotten to convert the status to a string before printing it out in an error message.

yagizzha, 2022-02-04:
[image]
now the first part isnt working ,its receiving initial state then gives error

flynneva, 2022-02-10:
@yagizzha could you include the command you ran in the screenshot?

Also, are you recompiling and sourcing the newly generated binaries with `source install/setup.bash`?

yagizzha, 2022-02-11:
im running it with 
"
cd dev_ws
. install/setup.bash
source /opt/ros/foxy/setup.bash
. ~/ros2_foxy/install/local_setup.bash
sudo chmod a+rw /dev/ttyUSB0
export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-11
ros2 topic list -t" 
i use that on all new terminals , then for grbl 
"
sudo chmod a+rw /dev/ttyUSB0

ros2 run grbl_ros grbl_node --ros-args --params-file /home/yagi/dev_ws/src/grbl_ros/config/cnc001.yaml

ros2 service call /spawn turtlesim/srv/Spawn "{x: 1, y: 1, theta: 0, name: 'turtle1'}"

"

flynneva, 2022-11-13:
Closing since I think this issue was resolved. Reopen it if it isnt

closed 2022-11-13 by flynneva