flynneva/grbl_ros #119: Is there a version for ROS1

blackhorsewu, 2022-05-16:


flynneva, 2022-05-16:
@blackhorsewu there might be somewhere out there but no nothing for ROS1 written by me for grbl...sorry 😅

Sehgaljazzy, 2022-09-10:
I'm using a Opentrons OT-1 pipetting robot which implemented using smoothie board ( a CNC controller) containing G-codes for it to move.

-Opentrons developed a Python API where I could send python scripts to control the robot.

-now I'm trying to create ROS infrastructure where I can connect smoothie-board with raspberry pi and install ros on it to achieve pipetting and add other modules like object detection, temperature control etc. Can we achieve this infrastructure with ros despite smoothie-board being a CNC machine? 

some links for background
-http://smoothieware.org/smoothieboard-v1
-https://www.youtube.com/watch?v=kBu9XAkfj_s

flynneva, 2022-09-10:
Hi @Sehgaljazzy! Friendly reminder to make new issues for new threads in order to keep things separate and help others find answers for what they are looking for 😁

As for using a "smoothie board" with grbl_ros, I've never used one or tested it but sure it looks like the smoothie boards support a "grbl" mode:

http://smoothieware.org/grbl-mode

Might be worth looking into and sounds fun!

closed 2022-11-13 by flynneva