flynneva/grbl_ros #120: [machine pos topic is not updating]

wildanxgifari, 2023-03-14:
Hi Flynn!

I enjoy your work very much, I use it to control my delta printer via joystick

I want to get feedback for the robot position
however echoing from /cnc_id/machine_position does not output anything

What may cause it to happen?

Thx, Wildan

flynneva, 2023-03-21:
Hey @wildanxgifari! Glad you enjoy this package. I'm traveling right now but will look into your issue when I can soon.

flynneva, 2023-03-28:
@wildanxgifari so this repo I believe uses actions, which feedback on the current position should be sent back to the action caller.

Let me double check that it's functional on my bench and get back to you.

glass3cat, 2024-09-15:
I also get the same position display when I use ros2 topic echo /cnc_001/machine_position.

~$ ros2 run grbl_ros grbl_node --ros-args --params-file ~/ros2_ws/src/grbl_ros/config/cnc001.yaml
[INFO] [1726366385.151330662] [grbl_device]: [ ? ] <Idle|MPos:0.000,0.000,0.000,0.000|Bf:15,127|FS:0,0>
[INFO] [1726366671.853200482] [grbl_device]: [ ? ] <Run|MPos:-0.011,0.011,0.000,0.000|Bf:14,127|FS:1018,0|WCO:0.000,0.000,0.000,0.000>
[INFO] [1726366671.873773668] [grbl_device]: [ ? ] <Run|MPos:-0.068,0.068,0.000,0.000|Bf:14,127|FS:1188,0>
[INFO] [1726366671.893868571] [grbl_device]: [ ? ] <Run|MPos:-0.214,0.214,0.000,0.000|Bf:14,127|FS:1527,0>

[INFO] [1726366672.956466212] [grbl_device]: [ ? ] <Run|MPos:-44.820,44.820,0.000,0.000|Bf:15,127|FS:0,0>
[INFO] [1726366672.978406569] [grbl_device]: [ ? ] <Run|MPos:-44.955,44.955,0.000,0.000|Bf:15,127|FS:0,0>
[INFO] [1726366673.002776816] [grbl_device]: [ ? ] <Idle|MPos:-45.000,45.000,0.000,0.000|Bf:15,127|FS:0,0|WCO:0.000,0.000,0.000,0.000>

~$ ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: G0 X-45 Y45}'

~$ ros2 topic echo /cnc_001/machine_position
position:
  x: -0.044988999999999994
  y: 0.044988999999999994
  z: 0.0
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
---
position:
  x: -0.045
  y: 0.045
  z: 0.0
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
---