flynneva/grbl_ros #126: Project status?

proffalken, 2024-12-18:
Hello,

I'm about to interface a robot arm with ROS2 that uses NEMA17 motors and before I go off and write my own firmware, I figured I'd see if I can reuse my Woodpecker CNC board.

This project looks ideal for what I want to achieve, but I've noticed that it hasn't been actively developed for a while.

Is that because it's complete, or because you've moved on to other things?

flynneva, 2024-12-23:
@proffalken I've just been busy with life 😅 this project definitely isn't complete yet but I'd be happy to help review and code some new features as needed.

Additionally I'd be happy to also just make you a co-maintainer in the project if you're serious about it. That way you don't really need me at all in the case you want to move fast.

Maybe first place to start would be testing it out on your setup, seeing what's missing and making issues for them. Once you do and it's clear to me you're on a good path I'd be happy to make you a maintainer and help anywhere I could.

flynneva, 2024-12-23:
@proffalken also it might make sense just starting from scratch in a new repo/project/branch and rewriting this whole thing since the code here now is quite old and I'm sure there are much better ways to do the same thing now.

proffalken, 2025-01-02:
Hey, thanks for the response, and I know exactly what it's like with life getting in the way!

My C++ isn't even close to good enough to take this on, but I can definitely test what's there and see if it works with Jazzy, and I'm learning C++ as one of my goals this year so I might be able to look at taking it on in future depending on how well that goes.

Good to know it's "not dead, just having a nap", I'll let you know how I get on!

proffalken, 2025-01-02:
(I've closed this because it doesn't need to be open, but I'll post more updates here or log new issues as an when I get to them)

closed 2025-01-02 by proffalken

proffalken, 2025-01-16:
Continuing here because I suspect people will probably read this issue.

I've got a Woodpecker CNC controller running Grbl 0.9j (yes, it could do with an upgrade!) and whilst grbl_ros talks to it fine and appears to send the expected commands, the response back in the grbl_ros console is as follows:

```
[WARN] [1737060901.471802881] [grbl_device]: Received status from machine does not have required fields
[WARN] [1737060901.472602387] [grbl_device]: Status should always include machine state and current position
[WARN] [1737060901.473363034] [grbl_device]: Received: %s
```

I'm not sure if this is a GRBL thing because it's such an old version, or just a configuration flag that I need to set somewhere, and it doesn't appear to affect the commands being carried out, I assume it just affects knowing where the effector is at any given time?

proffalken, 2025-01-17:
For anyone else looking to use this with a Woodpecker, the following commands helped:

```
# Unlock the CNC controller
ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: $X}'

# Set the unit to mm
ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: G21}'

# Set the distance to relative (so we add/subtract the required distance from the current position)
ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: G91}'

# Send the feed rate with the command:
ros2 action send_goal /cnc_001/send_gcode_cmd grbl_msgs/action/SendGcodeCmd '{command: G1 F500 Z-110}'
```