flynneva/grbl_ros #127: Integrating grbl_ros with Moveit2 etc.

proffalken, 2025-01-18:
@flynneva - just to let you know that I've got this up and working via the command line ('ve left notes on what I did on [the previous issue I opened](#126 )):

[!Video of robot arm working](https://github.com/proffalken/RobotArm/raw/refs/heads/main/Media/ArmWorkingWithGRBL_ROS.mp4)

What I'm struggling with is how I integrate GRBL_ROS with my existing controller so that when I move the arm in Moveit2, it sends the correct GCode to move the arm in the real world.

If there's any chance at all you could give me a hint that would be awesome - I've [asked on reddit](https://www.reddit.com/r/ROS/comments/1i44u6x/) but the answer I've had back hasn't helped!

flynneva, 2025-01-18:
@proffalken glad to hear you got it working 

How do you send commands to MoveIt2 to control the simulated arm? What does MoveIt2 output when you do?

flynneva, 2025-01-18:
You'll need a way to convert the same commands you use for MoveIt to GCODE, and then use that output to call the grbl_ros service

proffalken, 2025-01-18:
>How do you send commands to MoveIt2 to control the simulated arm? What does MoveIt2 output when you do?

Good question, I'm still learning! 😂 

Moveit2 appears to be based on RViz and computes the IK movements required to send the messages to the controller.

The controller then updates Gazebo in my digital twin, so I'm trying to work out how to get the controller to send the action commands to GRBL_ROS instead.

> You'll need a way to convert the same commands you use for MoveIt to GCODE, and then use that output to call the grbl_ros service

ok, so that's where I'm falling short - I'll dig into the previous basic controller that I wrote based on a Udemy course that I did and see if I can work out how to do that.

Appreciate you taking the time to reply, I'll let you know how I get on!

proffalken, 2025-01-18:
OK, I've worked out where I need to make the change, it's in https://github.com/proffalken/RobotArm/blob/main/workspace/src/robotarm_controller/src/robotarm_interface.cpp#L141-L182

Somehow I need to get this sending the goal to the action server instead of writing it to the USB port, let me see how that goes!

glass3cat, 2025-09-29:
I'm not sure if this will be helpful, but if you want to use ROS 2 Control's higher-level control functions (e.g., path planning with MoveIt!) while using grbl_ros nodes for the final output, the integration would look like this.

There is no need to develop a custom hardware interface like GRBL_CNC_Kinematics.

Instead, place an independent node (let's call it the "kinematics adapter node") that performs kinematics calculations and G-Code conversion between ROS 2 Control's JointTrajectoryController and the grbl_ros node.

Role of the Kinematics Adapter Node
Input: Subscribes to the target joint position topic published by JointTrajectoryController.

Calculation: Converts the subscribed joint angles into CNC working coordinates (X, Y) using forward kinematics.

Output: Creates a G-Code string containing the calculated X and Y coordinates (e.g., G1 X100.0 Y50.0 F500) and sends it as an Action Goal (/cnc_001/send_gcode_cmd) to the grbl_ros node.

In this configuration, ROS 2 Control is responsible for generating and managing angle commands, leaving direct communication to GRBL to grbl_ros. This makes it possible to control a GRBL device via ROS 2 Control without having to create GRBL_CNC_Kinematics yourself.